Spartronics4915/2015-Recycle-Rush #23: Versioning

TarkanAl-Kazily, 2015-01-28:
We need to create a versioning system that will allow us to track the changes made in each rendition of our code. While we have done a manual system within the code itself [in the past](https://github.com/Spartronics4915/2014-Aerial-Assist/blob/master/ArcadeDriveRobot/src/org/usfirst/frc4915/ArcadeDriveRobot/Version.java) where we manually edit the version number, it would be better if we had it so that either Eclipse or Github automatically increments the version number every time a milestone is reached. This could be at each commit, pull request or main branch merge.

Phroa, 2015-01-29:
git tags <http://git-scm.com/book/en/v2/Git-Basics-Tagging> may be a great
way to (or at least help) establish *general* version numbers in the source.

As far as getting that data into the code, doing a system call to `git
describe` displays tag version and the specific git commit hash. One
pitfall I see is that the robot may not have git binaries installed- I'm
totally unsure whether it does (not).

Phroa, 2015-01-29:
Addressed by pull request #26.

closed 2015-01-29 by Phroa