Spartronics4915/2015-Recycle-Rush #25: Driver Station Communication

TarkanAl-Kazily, 2015-01-28:
We need a good way of sending messages to the Driver Station, because of it's versatility in debugging and troubleshooting. Last year we made [this class](https://github.com/Spartronics4915/2014-Aerial-Assist/blob/master/ArcadeDriveRobot/src/org/usfirst/frc4915/ArcadeDriveRobot/SendUserMessages.java) which would allow us to send messages to the text box on the Smart Dashboard. We need to see if that can be used or improved for this year. If not, can we do something else to communicate with the drivers?

RonnieNigash, 2015-02-06:
@TarkanAl-Kazily @Overwatch301 @Phroa 

I looked into migrating code from last year's team, looks as though that won't be the course.
Last year we used the DriverStationLCD class, recent updates have rendered it obsolete. 

"The User messages window on the Driver Station has been removed, therefore this functionality (DriverStationLCD class) has been removed from WPILib. As a replacement, the default Dashboard contains a number of controls and indicators of different types on the "Basic" tab that teams may wish to use to display robot information or provide input. To write to these controls, use the SmartDashboard class, the key names to use are listed on the Dashboard."

This link should point you in the right direction. :+1: 
http://wpilib.screenstepslive.com/s/4485/m/24192/l/272692?data-resolve-url=true&data-manual-id=24192

TarkanAl-Kazily, 2015-02-06:
Thanks! Yeah, there have been a lot of changes. The thing with the smart dashboard this year is that there are multiple kinds, one for each language that we use for programming our robot. So it seems like from the link Ronnie gave us, we should be able to use the Labview Dashboard with our Java robot and get these extra features from it.

TarkanAl-Kazily, 2015-02-17:
DriverStation.reportError("STRING HERE", false) does the trick.

closed 2015-02-17 by TarkanAl-Kazily