Spartronics4915/2015-Recycle-Rush #83: Check for 'null' object

binnur, 2015-02-08:
To ensure we are not creating null exceptions, we need to ensure we check for 'null' before accessing the objects. Currently, OI.java has the following commented as limit switches, etc. are not wired. However, anything that may get disconnected/broken during the competition needs to be tested with 'null' before calling. 

Wrap the following calls with: if (RobotMap.limitSwitchTop != null) then add LiveWindow.... 
        // LiveWindow.addSensor("Elevator", "Upper limit switch", RobotMap.limitSwitchTop);
        // LiveWindow.addSensor("Elevator", "Lower limit switch", RobotMap.limitSwitchBottom);
        // LiveWindow.addSensor("Elevator", "Linear potentiometer", RobotMap.potentiometer);

Overwatch301, 2015-02-09:
OI.java and RobotMap.java now both have null checks before every possible physical part method call.

binnur, 2015-02-09:
I'll close this issue - open a new one for the vision code. Thanks!

closed 2015-02-09 by binnur